APS-2BM-MIC/ipython-user2bmb #8: Suggestion: avoiding calling RE inside a function

danielballan, 2018-02-07:
We advise you to avoid calling `RE` from inside a function, as is done [here](https://github.com/APS-2BM-MIC/ipython-user2bmb/blob/master/profile_bluesky/startup/82-mona.py#L115). We can see this for demo purposes, but it's still good to avoid having this things floating around, especially in an interactive namespace. They can lead to unexpected and confusion behavior; the RunEngine should always be called directly. It's better to use `functools.partial` to make a succinct plan:

Before:

```python
def demo():
    RE(mona_core(detectors, adsimdet.cam.acquire, num=3), md=metadata)
demo()
```

After:

```python
import functools
demo = functools.partial(mona_core, detectors, adsimdet.cam.acquire, num=3, md=metadata)
RE(demo())
```

where `mona_core` must be modified to accept an `md` argument.

prjemian, 2018-02-07:
I agree.

prjemian, 2018-02-10:
will resolve as part of #3

commit 3b419a9e (2018-02-11): fixes #8
closed 2018-02-11 by prjemian