APS-2BM-MIC/ipython-user2bmb #10: problem counting with ScalerCH

prjemian, 2018-02-10:
per #7, switched `EpicsScaler` to `ScalerCH` but that produced a problem when using `RE(bluesky.count([scaler]))`

where

    scaler = ScalerCH("xxx:scaler", name="scaler")

danielballan, 2018-02-12:
@tacaswell Any guidance here? Looks like this is typing in a jsonschema validation error when the documents are emitted, likely because `ScalerCH` is generating invalid (blank?) data key names.

tacaswell, 2018-02-12:
The issue is that the `name` PV is an empty string so when (on init) the channels asks what it's name should be, it happily takes the empty string.  

Either put names in and then re-call `match_names` or remove channels without names from the read attrs

danielballan, 2018-02-12:
Hmm. It has _a_ name -- `name="scaler"`. Do the channels not name themselves by default? It seems like `ScalerCH` should blow up at init time if it has names/read_attrs that will generate invalid documents.

tacaswell, 2018-02-12:
In the `__init__` of the individual channels they call `self.match_name` which sets the channel's `.name` to what ever is in the name PV (which out the parent's prefix).

danielballan, 2018-02-12:
OK, the key fact I was missing here is that scaler IOCs have a concept of names, which they expose through PVs. ScalerCH tries to account for that and sync to _ophyd_ names to the names known to the IOC. This has not come up in any other ophyd Devices, because IOCs don't typically encoded user-readable names. In that context, I can interpret Tom's original comment:

> Either put names in and then re-call match_names or remove channels without names from the read attrs

"Put it names" means put a string to the IOC. Then, sync the ophyd Device to that name using `match_name`. As a comment in `ScalerChannel` indicates, we could do that second step automatically by monitoring the IOC's name PV.

danielballan, 2018-02-12:
See https://github.com/NSLS-II/ophyd/pull/509 which omits nameless channels from the `read_attrs` at init time.

tacaswell, 2018-02-13:
@prjemian did you get this sorted?

prjemian, 2018-02-13:
No, but not important now

prjemian, 2018-02-13:
All that needs to be done is to add labels to each desired scaler channel
in EPICS?

On Feb 13, 2018 5:02 PM, "Pete Jemian" <prjemian@gmail.com> wrote:

No, but not important now

danielballan, 2018-02-13:
Yes, seems so.

prjemian, 2018-02-14:
Tested this just now.

This works for me *iff* the channel names (.NM1, .NM2, ...) are defined *before* the call to `ScalerCH()`.  If the names are changed after this call, the ophyd object does not learn about the new name.

Also, should `read_attrs` and/or `configuration_attrs` be populated initially with the channels that have names?

tacaswell, 2018-02-14:
You can call `sclr.match_names()` to re-sync them.  Maybe that should be done in the `stage` method?  We could also install a monitor onto the name PVs, but then they could change mid-scan which would break _lots_ of things.

NSLS-II/ophyd#509 filters to only channels with names.

prjemian, 2018-08-03:
This has a [workaround](https://github.com/APS-USAXS/ipython-usaxs/issues/34#issuecomment-407783288) until ScalerCH is fixed up.

closed 2018-08-03 by prjemian