APS-2BM-MIC/ipython-user2bmb #31: theta

decarlof, 2018-06-13:
the rotation axis position is currently not saved in the hdf file.

decarlof, 2018-06-14:
this is correct for the streaming, still we want to save theta in the hdf file. Currently this is not happening,

f

prjemian, 2018-06-15:
In the attributes XML file, there is this entry:

```
    <Attribute name="SampleRotary"
        type="EPICS_PV"
        source="2bmb:m100.RBV"
        dbrtype="DBR_DOUBLE"/>
```

This is describedin the layout file:

```
    <dataset name="sample_rotary" 
            source="ndattribute" 
            ndattribute="SampleRotary" 
            when="OnFileClose"> 
            <attribute name="units" source="constant" value="mm" type="string"></attribute>
    </dataset>
```

A single value for this PV ends up in the data file at this HDF5 address: `/measurement/instrument/sample/setup/sample_rotary`

with this content in one of the HDF5 files I examined:

```
          sample_rotary:float64 = 9087.6855
            @units = mm
            @NDAttrName = SampleRotary
            @NDAttrSourceType = NDAttrSourceEPICSPV
            @NDAttrSource = 2bmb:m100.RBV
```

Looks like the 2bmb:m100.EGU field is wrong.  Should be `degrees`.  Easy to fix for the future.

decarlof, 2018-06-15:
/measurement/instrument contains motor positions at a file close. It is used to get a snapshot of static motor position at that moment. For the motors involved in the scan (in our case only the sample rotary theta) we also save the list of all angles associated to each projection under \exchange\theta. This is required to do the reconstruction off line (using the hdf file).

 \exchange\theta in missing from the file generated during the mona session

aglowacki, 2018-06-15:
We used to have a theta under /exchange that would save it per frame but you would have 0's for the pre dark and pre white and 180's for post dark and post while. This was unwanted so we moved the logic to the scan script so that it collects an array of theta's and appends it to the hdf5 file when the scan is over.

prjemian, 2018-06-15:
Good.  Bluesky can do similar.  AD writes the HDF5 file.  When that is complete, Bluesky will re-open the file and store the array of computed theta values based on the math above.

decarlof, 2018-06-15:
>>> Looks like the 2bmb:m100.EGU field is wrong

thank you for catching this. I update the xml at [dxfile docs](http://dxfile.readthedocs.io/en/latest/source/demo/doc.areadetector.html)

decarlof, 2018-06-15:
that would be ok for now as we are only taking white/dark at the beginning and at the end but in general when we take white/dark every so many projections we should save the theta at which white/dark were taken in theta_dark and theta_white vector (see [docs](http://dxfile.readthedocs.io/en/latest/source/xraytomo.html#exchange). This was we can always use the proper white/dark when normalizing.

prjemian, 2018-08-04:
What HDF5 address to use?  this: `/exchange/theta`

Also needs `units` attribute

prjemian, 2018-08-04:
# Consequence

If Bluesky is to update the HDF5 data file, it needs write access to that file system.

For the PointGrey detector at 2-BM-B, data is written to the `/local` partition which is only accessible from *lyra*.  That means Bluesky has to run on *lyra*.

on *arcturus*:

```
182.16s - flying   x = 0.0000  y = 0.0000  r = 180.5000  theta = 180.5000  image # 1500
Could not find /local/user2bmb/mona/2018/08/03/ce890410-cdae-4109-806c_000000.h5
Cannot add /exchange/theta
Out[1]: ('97c765e0-4d50-426b-abf3-df30c948ac96',)                                                                                                                                                      
```

prjemian, 2018-08-04:
Here is from a successful data set:

```
    theta:float64[1500] = [0.0, 0.12008005336891261, 0.24016010673782523, '...', 180.0]
      @units = degrees
      @description = computed rotation stage angle
```

commit c58a8927 (2018-08-04): fixes #31 - thanks @decarlof !!!
closed 2018-08-04 by prjemian