APS-2BM-MIC/ipython-user2bmb #35: optimize rotary stage speed calculation for fly scan

prjemian, 2018-08-18:
We can rotate the sample stage faster than the current calculation provides.  The user wants to specify `acquire_time` (exposure time for each image frame) based on signal and noise.  We want o minimize the angular blur that is a consequence of acquiring an image frame while the sample stage is rotating.  But, we want to acquire a full tomo dataset as fast as possible.

How much blur is acceptable?  A good starting estimate is no more than 0.1 pixel/frame.

The algorithm is:

* user specifies `acquire_time` (based on signal and noise)
* assign blur of 0.1
* calculate rotation speed, limited to 0.5 <= speed < 30 degrees/s
* calculate blur with constrained speed

prjemian, 2018-08-18:
can solve this with no beam, move to later milestone

prjemian, 2018-08-18:
@decarlof : deposit example code here

decarlof, 2018-08-18:
@prjemian [here it is](https://github.com/decarlof/util/blob/master/txm/fly_scan.py)

prjemian, 2018-08-18:
@decarlof : Good, I'll use those functions directly (as can be) in the mona/bluesky code

commit b9433f9a (2018-08-19): fixes #35, fixes #37
closed 2018-08-19 by prjemian