APS-2BM-MIC/ipython-user2bmb #37: drop images at rotation speed other than 1 deg/s,

prjemian, 2018-08-18:
Relates to #35 

Want to optimize rotation speed but discover that we only get 750 images when speed is 2 - 18 degrees second (highest speed that we expect to use).

How to get full number of frames when scanning at different rotational axis speeds?

prjemian, 2018-08-18:
* 1 deg/s, 0.150 acquire time only gets 750 images

prjemian, 2018-08-18:
related to #36?

prjemian, 2018-08-18:
* 1 deg/s and 0.1 s acquire time is in the Goldilocks zone.  Why?

prjemian, 2018-08-19:
* 1.05 deg/s, 0.1 acquire time gets 1500 images
* 1.1 deg/s, 0.1 acquire time gets 1500 images
* 1.2 deg/s, 0.1 acquire time gets half of 1500 images

prjemian, 2018-08-19:
Even without Bluesky, just operating from GUI screens, 1.2 deg/s gets only half the expected images

Check the detector setup in bluesky code and compare with non-bluesky operations code

prjemian, 2018-08-19:
@decarlof post the file here

decarlof, 2018-08-19:
[PG_camera.zip](https://github.com/APS-2BM-MIC/ipython-user2bmb/files/2300063/PG_camera.zip)

prjemian, 2018-08-19:
Nothing obviously different in these files.  PG is in Overlapped mode.  The file `tomo_scan_lib.py` calls SoftwareTrigger directly several times.  No additional config details to be learned.

prjemian, 2018-08-19:
Running the camera from the GUI, at 0.005 s acquire_time (and 0.008 s acquire period), with internal trigger and 1000 multiple exposures and HDF5 save enabled, the maximum frame rate was 128 fps.

When running the fly scan from the GUI (no bluesky in use), the maximum frame rate was 10 fps.  When adjusting the rotation speed faster, the frame rate drops.

We suspect that the camera is not getting the triggers from the motor controller.  Something is wrong:

* with our setup of the cmaera
* out setup of the fly scan software (such as scanDelta calculation)
* with the motor controller trigger output

Will use camera at 1 deg/s for now and further diagnose/debug this during the shutdown.

prjemian, 2018-08-19:
use acquire_time < 0.1 to get more than 10 fps

commit b9433f9a (2018-08-19): fixes #35, fixes #37
closed 2018-08-19 by prjemian