APS-2BM-MIC/ipython-user2bmb #53: need test mode for shutters

prjemian, 2018-11-03:
i am wondering if you were able to implement a testing mode for the bluesky at 2-bm-b (so we can run without operating the shutter and ignoring current level); Andrew would like to run some tests without interfering with ops in A

prjemian, 2018-11-08:
These are the PVs (and record types) to simulate:

```
[user2bmb@lyra,44,~]$ caget 2bma:A_shutter:{open,close}.RTYP PA:02BM:STA_A_FES_OPEN_PL
2bma:A_shutter:open.RTYP       bo
2bma:A_shutter:close.RTYP      bo
PA:02BM:STA_A_FES_OPEN_PL      OFF
```

In the simulation, the status PV should change a short time after one of the two action PVs has been changed.  Each action PV is nominally at value zero (no string value assigned).  When changed to one, it returns back to zero *shortly*.  Assume that time is ca. 0.5 seconds.

prjemian, 2018-11-09:
moved issue to APS_BlueSky_tools repo

decarlof, 2018-11-16:
@prjemian has this been addressed? we are ready to use 2-BM-B in test mode. Also where is the APS_BlueSky_tools repo?

prjemian, 2018-11-17:
latest release:  0.0.34
repo: https://github.com/BCDA-APS/APS_BlueSky_tools
docs: https://aps-bluesky-tools.readthedocs.io/en/latest/
PyPI: https://pypi.python.org/pypi/APS_BlueSky_tools
conda: not producing for conda now, will do this with first release 1.n.n

prjemian, 2018-11-17:
I believe this was not yet implemented at beam line so will re-open.  All it needs to be useful now is a Signal (hopefully from an EPICS PV) that indicates which station, if any, is using the beam.  The other station will use simulators or bypasses to enable software testing.

Worth re-opening this issue then.

I propose an EPICS mbbo (multi-choice) record.  Instrument team sets this manually until someone can propose a reliable algorithm to set it automatically.

value | string
---- | ----
0 | unassigned
1 | 2-BM-A
2 | 2-BM-B

This will be very easy to implement, could go into *any* existing IOC.  Best belongs in an IOC under regular maintenance, such as for 2bma: or 2bmb.  I'll ask Mark E. to add this on Monday to 2bma:

The basic idea is applicable to any beam line.  The mbbo record has capability for up to 15 different values in addition to the zero default.

Thoughts?

prjemian, 2018-11-19:
@Engbretson - We talked about this and concluded NOT to put this in either A or B station's IOCs.  I'll add this PV as described in #55.  Thanks for the discussion.

prjemian, 2018-11-19:
Work on this issue should follow completion of #55.

prjemian, 2018-11-19:
Summarizing, work on them in this order:  #54, #55, then #53

Engbretson, 2018-11-19:
Could always be shifted to one of them after all the testing is done, if required.  May need more fake support stuff before this is done.

Engbretson, 2018-11-19:
Yeah, existing simulation mode in epics seems to be an all or nothing thing. Not something which allows for a single ‘live’ environment and everyone else running in ‘simulation’.  I remember when the PSS controls worked that way on multi station sectors . . . to lock out anyone else twiddling with shutters, but they stopped doing that long ago.

prjemian, 2018-11-20:
@decarlof : With this new code, we can configure items just in operations or test modes, as decided when result from `using_beam_in_2bmb()` is `True` or `False`, respectively.  Tomorrow (Tuesday)?

prjemian, 2018-11-20:
@cpchuang, @decarlof - Tomorrow (Tuesday) morning at 2-BM-B?

commit f45704b9 (2018-11-20): fixes #53 configure for operations or test mode
closed 2018-11-20 by prjemian