APS-2BM-MIC/ipython-user2bmb #54: add standard softIoc management script for "mona:" IOC

prjemian, 2018-11-17:
For the MONA project, we need a few PVs so the MONA software pipeline can provide feedback to the running scan.  Presently, this PV `mona:StopAcquisition` is important to advise the bluesky plan to stop (`RE.stop()`) the current scan.  This PV is used as a bluesky suspender which means if the IOC is not running, the plan won't start.  This error:

```
<class 'str'>: (<class 'TimeoutError'>, TimeoutError('Failed to connect 
to mona:StopAcquisition',))
```

The IOC is started using these commands:

```
cd ~/.ipython/profile_2bmb/startup/
screen
softIoc -d mona.db
^A d   (press Ctrl and A keys together, then d key to leave the screen session running in the background)
```

BCDA has a `manage.sh` shell script to manage a soft IOC such as this, providing simpler commands such as:

* `./manage.sh start`
* `./manage.sh stop`
* `./manage.sh restart`
* `./manage.sh checkup`   (starts soft IOC if cannot find expected one running)
* `./manage.sh console`    (provides access to IOC's console)

This script can be placed in a cron file so that the IOC will start automatically after a reboot of the workstation.

Can set this up Monday.

cpchuang, 2018-11-19:
It's a good idea to start soft IOC everytime the machien is rebooted. Perhaps a error handeling to remind the restart  if this IOC is not responsive for any reason.

prjemian, 2018-11-19:
Start off with moving existing parts into a new `iocmona` directory.  Monitoring (for start or restart) is part of the standard `manage.sh` shell script.  This is automated by adding as a `cron` task for periodic review (default is every 5 minutes).  Could run as often as 1 minute intervals but that seems excessive.  As a cron task, this will run (and start the IOC) within 5 minutes of a machine reboot.

prjemian, 2018-11-19:
Here's an example of [`manage.sh`](https://git.aps.anl.gov/bcdaioc/2bmbPG3/blob/master/iocs/pointGreyIOC/iocBoot/iocPointGrey/2bmbPG3.sh) (requires APS credentials to view).

prjemian, 2018-11-19:
Another example of [`manage.sh`](https://github.com/epics-modules/xxx/blob/master/iocBoot/iocxxx/softioc/xxx.sh) from the synApps [`xxx`](https://github.com/epics-modules/xxx) open-source repository.

prjemian, 2018-11-19:
.sh needs a checkup feature (re)added

prjemian, 2018-11-19:
# USAGE

```
user2bmb@lyra ~/.../startup/ioc2bmbmona $ /home/beams/USER2BMB/.ipython/profile_2bmb/startup/ioc2bmbmona/2bmbmona.sh usage
Usage: /home/beams/USER2BMB/.ipython/profile_2bmb/startup/ioc2bmbmona/2bmbmona.sh {start|stop|restart|status|checkup|console}
```

closed 2018-11-19 by prjemian

prjemian, 2018-11-19:
## periodic task : cron

runs on same workstation and account as IOC (user2bmb@lyra)

```
user2bmb@lyra ~/.../startup/ioc2bmbmona $ crontab -l
#=====================
# call periodically (every 5 minutes) to see if IOC is running
#=====================
#	   field	  allowed values
#	   -----	  --------------
#	   minute	  0-59
#	   hour 	  0-23
#	   day of month   1-31
#	   month	  1-12 (or names, see below)
#	   day of week    0-7 (0 or 7 is Sun, or use names)

# check every five minutes if IOC is running in screen session
*/5 * * * * /home/beams/USER2BMB/.ipython/profile_2bmb/startup/ioc2bmbmona/2bmbmona.sh checkup 2>&1 > /dev/null
```

commit a22defc7 (2018-11-19): fixes #54